pbatard/EfiFs #4: UFS drivers with rEFInd?

saghm, 2016-11-07:
Hi! I was wondering if you know whether it's possible to use the UFS drivers from here with the rEFInd booloader. I tried downloading the latest ones and dropping them into rEFInd's drivers directory, but when I boot into refind, it shows the message "Load error returned from ufs(1|2)_x64.efi" for each of them.

pbatard, 2016-11-07:
The UFS driver should work. It was tested as a standalone driver with various UEFI firmwares (NB: I have a unit test for each driver in efifs, which I conduct for IA32, x64 and ARM). Especially I'm not aware of any issues when loading the driver.

What happens when you load the driver from the UEFI shell (`load ufs2_x64.efi`)? If that loads then it means the issue is probably with rEFInd.
Also, what version of rEFInd are you using? I seem to remember that Rod had an issue with loading the drivers had some stage, which he since fixed. So please make sure you use the latest.

saghm, 2016-12-03:
Sorry about the late response! I'm using 0.10.4 (the latest version provided by my package manager), which appears to the same version provided by his website.
When I try to load it in the shell, I get the following error:

```
load: Image fs0:\EFI\BOOT\drivers_x64\ufs(1|2)_x64.efi error - Load Error
```

Unfortunately this doesn't seem to provide any more detailed information than when rEFIfind fails to load it. Any chance you have some other suggestions to help with debugging this? (If not, that's totally fine, and thanks for the help!)

pbatard, 2016-12-06:
What system are you using exactly?

I don't see any issue loading the drivers in either the QEMU shell or the (x64) shell of an actual intel NUC system. I also haven't received any reports, besides yours, that the drivers provided couldn't load in an UEFI shell.

So it looks to me like it's your UEFI firmware or shell that may be at fault. have you tried on a different UEFI system? What was the outcome there?

The only thing I can think of, in case you are using HP hardware, is that some of their UEFI firmwares have been poorly designed and try to keep exclusive access on all partitions (which effectively prevents a driver from being able to service such partition). However, rEFInd does have code to work around that specific bug when loading a driver, so you should have been able to use the UFS drivers there.

Unfortunately, it seems that there exist a few hardware manufacturers out there that have taken some "liberties" with the UEFI specs in their firmware, especially when it comes to older firmwares, so I'd tend to fault the hardware manufacturer if the problem only occurs with a specific type of machine.

I sure wouldn't mind hearing the exact make/manufacturer of the machine you have an issue with, as well as the UEFI version provided by its firmware though...

saghm, 2016-12-08:
I'm using a Lenovo Ideapad 700 (not the Y700, which seems to always be the only thing that comes up when I google it). According to the EFI shell, the firmware version is 2.4.

Given your bewilderment that the shell wasn't able to load the driver, I figured I'd try to get a different shell from somewhere to see if that worked. I found one on the Arch wiki that was able to load the drivers fine, so I guess the old shell I had was just not very good (I wish I could remember where I downloaded it from, but it was a while back).

Unfortunately though, the shell is just something that I dropped into the rEFInd "tools" directory to be able to go into when I booted, so having the shell working doesn't have any impact on rEFInd's ability to load the driver (I tried just in case, but no luck).

This looks to be an issue with rEFInd and not the drivers, so I'm guessing there's not much you can do on your end. Thanks for the help!

pbatard, 2016-12-16:
Since I don't believe there's much more I can do, I will close this issue.
I will keep my eyes open for similar reports though.

closed 2016-12-16 by pbatard