pbatard/EfiFs #5: Can an ISO not be mounted via the iso9660.efi driver?

AdrianKoshka, 2017-03-11:
Pardon me for my ignorance on `UEFI`, but I was wondering if it would be at all possible via the `iso9660.efi` driver to `mount` an `.iso` file.

pbatard, 2017-03-11:
No. The driver only provides file system access capabilities, __not__ virtual partition capabilities, which is what you would also need.

This is just like Linux, really. If you want to mount a `.iso` file on Linux you need:
1. Your kernel to have an `ISO9660` or `UDF` driver, either as a native driver (i.e. already embedded in the kernel) or as a module. And you should consider that efifs's `iso9660.efi` driver is the equivalent of a kernel module, but for the UEFI "kernel".
2. A virtual device that will make your system consider the ISO image as a physical drive/partition, which you'd usually create with something like `losetup /dev/loop0 /path/to/your.iso`. __This is a part that neither the efifs driver or the UEFI system provide__.
3. A utility that will map the device/virtual device and make it accessible to the system. This is either `mount` on Linux or `map` on UEFI.

Most modern version of Linux' `mount` usually hide step 2 with a `-o loop` option, but what this does is call `losetup` behind the scenes, as step 2. is mandatory for mounting an image.

So without the part in 2, you can simply not "mount" an ISO on UEFI. And since 2. can not be provided by a file system driver, and, to the best of my knowledge, there also doesn't exist an equivalent UEFI shell utility, no, you will not be able to mount ISO images from the UEFI shell.

But of course, since UEFI is Open Source, you are very welcome to write such a utility, and propose it for inclusion in the next version of the UEFI shell.

closed 2017-03-11 by pbatard

paigethylamine, 2018-12-29:
thats pretty stupid

bam80, 2020-02-16:
> And since 2. can not be provided by a file system driver

@pbatard do you mean "losetup" can't be provided by UEFI driver at all?

pbatard, 2020-02-16:
Yes, because a file system driver is limited to just that, a driver that provides translation for UEFI to access a file system.

There is a very specific interface (or API if you want to call it that way) for UEFI file system drivers, that restricts them to implement calls for `Open()`, `Close()`, `Read()`, `Write()` and so on, and __none__ of these calls have the ability to magically make a virtual partition or virtual device appear. Instead, they very much expect the UEFI system to already have knowledge of the virtual device/virtual partition __before__ calls to `Open()`, `Close()`, `Read()`, `Write()` and so on are made, because these calls pretty much translate to _"Hey, can you please do some file I/O for file X __on device Y__"_, so you need _"device Y"_ to be known by the system before you place these calls. As I tried to explain above, this means you need a __SEPARATE DRIVER__, that implements a UEFI virtual disk device to translate calls to read or write a block of data from virtual UEFI disk device Y to reading or writing the relevant section of the file that disk device maps it to, so that you can then tell the system that you want to use the ISO9660 or UDF file system driver on that disk.

In other words, a file system driver needs an underlying disk device to already exist before it can perform file I/O. It can't just "invent" a device out of thin air, no matter how much you would like it to be able to do so...

So, once again, without someone writing a virtual device driver to map an ISO to a UEFI disk device (equivalent to what `losetup` does), so that you can add a virtual UEFI disk device to the system, you won't get easy access to ISO files from UEFI, even if you have an ISO9660 or UDF UEFI File System driver. That's because it'd be equivalent to telling the system _"Please open `some_dir/some_file.txt`"_ and the system telling you _"Well, I'd like to do that, but first you have to tell me on which mapped UEFI disk device `some_dir/some_file.txt` resides"_, which of course you can't since there isn't any disk device that maps to your ISO.

If you or anyone else is annoyed by this situation, then you'll need to find someone (not me) to write a `losetup` equivalent for UEFI because that's the only way you're going to be able to "mount" an ISO in an UEFI environment.

bam80, 2020-02-16:
@pbatard thanks, actually my question referred to any driver, not necessary filesystem one.
So it still needs to be a driver, or it may be implemented in UEFI Shell entirely?

It's a little pity you can't try it, I don't know whom that request may be most properly addressed to :)

pbatard, 2020-02-16:
> So it still needs to be a driver, or it may be implemented in UEFI Shell entirely?

It needs to be a driver since it needs to implement the disk I/O protocol. That's how UEFI is designed. Now, technically, you could probably shove that protocol into the shell if you really wanted to, but if you're going to do that, you might as well do it properly and write a driver that won't be tied to using __your__ custom specific version of the shell (that the EDK2 folks will never accept in the official repo, as opposed to a virtual disk driver, which they probably would).

Besides, if you are going to use the Shell, I really fail to see why you wouldn't be able to run `load virtual_disk_driver.efi` to instantiate the driver. There's no benefit whatsoever in trying to implement a virtual disk functionality as anything but a UEFI driver...