pbatard/EfiFs #6: Hi, How do you appy those patches??

davidm71, 2017-09-07:
Trying to compile your Efifs driver in VS2017 and copied latest gnu-efi to the gnu-efi folder, copied grub v2.02 to the grub folder (also did a git clone) and when I compile I get a bunch of type conversion errors. Must because I need to apply those patches? On my system TortoiseSvn opens those patch files and when I direct it to the grub folder it errors saying 'non working directory'.  Must not be correct way to apply the patches? Thanks

pbatard, 2017-09-07:
> and copied latest gnu-efi to the gnu-efi folder, copied grub v2.02 to the grub folder

That's most likely your issue.

As mentioned in the README, you need to use `git submodule init` and `git submodule update` to fetch the grub and gnu-efi directories. Copying them manually will __not__ work.

Once you have done that, you should be able to apply the patches.

Please make sure you follow the README instructions.

davidm71, 2017-09-07:
I did do just that typing git submodule init and git submodule update though had difficulty in that it would only work after typing 'git init' then the other two commands worked. I'll try again. Also is this the correct way to apply the patch?
git apply --ignore-space-change --ignore-whitespace --whitespace=nowarn file.patch

Thanks

davidm71, 2017-09-07:
Otherwise if I don't type 'Git init' I get this:
fatal: Not a git repository (or any of the parent directories): .git

pbatard, 2017-09-07:
1. If you are using Windows, then I'd recommend you use TortoiseGit. It has menus for submodules
2. Your need for `git init` is very suspicious. Something tells me you forgot to `cd` to the `efifs` directory after cloning it (in which case you will of course have loads of issues). Make sure you are in `efifs` before applying the `git submodules` commands
3. git apply should work. But again, If you are on Windows, I would recommend using TortoiseGit, then copying the patches into the grub subdirectory and select Apply patch serial
4. I would strongly suggest you familiarize yourself with git, as the issues you are having seem to have to do with not knowing how to use git, and it's not really my job to teach you that...

davidm71, 2017-09-07:
1. Yes I am in windows 10 64 and I also took your advice and installed TortoiseGit which recommended 'Git for Windows' or which I already had it installed.
2.  I definitely CD'd into the efifs folder before typing git submodule init and getting that error.
3. Will try 'Apply patch serial'
4.  True and I'm trying to get the hang of git but you could have more detail in your readme.

Thanks.

davidm71, 2017-09-07:
Finally got it working. Turns out you cant download the efifs package as a zip or tar file and expect that 7-zip or Winzip or WinRar will unpack the files like they should, or perhaps to run 'git submodule init' you need to git clone https://github.com/pbatard/efifs.git manually or else it won't work. Besides that 'Tortoisegit' utility update patch serially like you said worked like a charm. Now I can finally have lunch!

Thanks for your help. Appreciate it! 
Thanks Pete!

closed 2017-11-06 by pbatard