pbatard/EfiFs #10: error: "ARRAY_SIZE" redefined [-Werror]

ElvishJerricco, 2018-11-25:
Trying to follow the instructions at https://github.com/pbatard/efifs#bonus-commands-to-compile-efifs-using-edk2-on-a-vanilla-debian-gnulinux-91

I booted up a vagrant box with Debian 9.6 and ran this script:

```bash
#!/usr/bin/env bash

set -ex

sudo apt install nasm uuid-dev build-essential git

git clone --recurse-submodules -b edk2-stable201811 https://github.com/tianocore/edk2.git
git clone --recurse-submodules -b v1.3 https://github.com/pbatard/efifs.git
cd edk2
ln -s ../efifs EfiFsPkg
make -C BaseTools/Source/C -j 8 -l 8
source edksetup.sh
./EfiFsPkg/set_grub_cpu.sh X64
build -a X64 -b RELEASE -t GCC5 -p EfiFsPkg/EfiFsPkg.dsc -n 8
```

And I got this error:

```
"gcc" -g -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -ffunction-sections -fdata-sections -include AutoGen.h -fno-common -DSTRING_ARRAY_NAME=udfStrings -m64 -fno-stack-protector "-DEFIAPI=__attribute__((ms_abi))" -maccumulate-outgoing-args -mno-red-zone -Wno-address -mcmodel=small -fpie -fno-asynchronous-unwind-tables -Wno-address -flto -DUSING_LTO -Os -Wno-unused-but-set-variable -Wno-unused-const-variable -DFORMAT=efi-app-x64 -Os -DCPU_X64 -DGRUB_FILE=__FILE__ -DDRIVERNAME=udf -DDRIVERNAME_STR=\"UDF\" -DMDEPKG_NDEBUG -DDISABLE_NEW_DEPRECATED_INTERFACES -c -o /home/vagrant/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Udf/OUTPUT/src/grub.obj -I/home/vagrant/edk2/EfiFsPkg/grub/grub-core/fs -I/home/vagrant/edk2/EfiFsPkg/grub/grub-core/kern -I/home/vagrant/edk2/EfiFsPkg/src -I/home/vagrant/edk2/EfiFsPkg/EfiFsPkg -I/home/vagrant/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Udf/DEBUG -I/home/vagrant/edk2/EfiFsPkg -I/home/vagrant/edk2/EfiFsPkg/grub/include -I/home/vagrant/edk2/EfiFsPkg/grub -I/home/vagrant/edk2/EfiFsPkg/grub/grub-core/lib/minilzo -I/home/vagrant/edk2/MdePkg -I/home/vagrant/edk2/MdePkg/Include -I/home/vagrant/edk2/MdePkg/Include/X64 -I/home/vagrant/edk2/ShellPkg -I/home/vagrant/edk2/ShellPkg/Include -I/home/vagrant/edk2/IntelFrameworkModulePkg -I/home/vagrant/edk2/IntelFrameworkModulePkg/Include /home/vagrant/edk2/EfiFsPkg/src/grub.c
In file included from /home/vagrant/edk2/EfiFsPkg/src/grub.c:22:0:
/home/vagrant/edk2/EfiFsPkg/grub/include/grub/misc.h:35:0: error: "ARRAY_SIZE" redefined [-Werror]
 #define ARRAY_SIZE(array) (sizeof (array) / sizeof (array[0]))

In file included from /home/vagrant/edk2/MdePkg/Include/Uefi/UefiBaseType.h:20:0,
                 from /home/vagrant/edk2/MdePkg/Include/Uefi.h:23,
                 from /home/vagrant/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Ufs/DEBUG/AutoGen.h:16,
                 from <command-line>:0:
/home/vagrant/edk2/MdePkg/Include/Base.h:1317:0: note: this is the location of the previous definition
 #define ARRAY_SIZE(Array) (sizeof (Array) / sizeof ((Array)[0]))
```

This differs from the suggested commands only in that I am not building for arm/aarch64, and I checked out the latest releases of EfiFs and EDK II rather than master. However, switching to master yields the same result. I tried patching the grub header, but that just led to more of the same style of errors, so I gave up. I figure there must be something else wrong, or else no one would have been able to build this.

pbatard, 2018-11-25:
Not getting that error on my machine, using latest edk2 (from git):
```
root@vm-debian9 /usr/src # cd edk2
root@vm-debian9 /usr/src/edk2 # ln -s ../efifs EfiFsPkg
root@vm-debian9 /usr/src/edk2 # source edksetup.sh
WORKSPACE: /usr/src/edk2
EDK_TOOLS_PATH: /usr/src/edk2/BaseTools
CONF_PATH: /usr/src/edk2/Conf
Copying $EDK_TOOLS_PATH/Conf/build_rule.template
     to /usr/src/edk2/Conf/build_rule.txt
Copying $EDK_TOOLS_PATH/Conf/tools_def.template
     to /usr/src/edk2/Conf/tools_def.txt
Copying $EDK_TOOLS_PATH/Conf/target.template
     to /usr/src/edk2/Conf/target.txt
root@vm-debian9 /usr/src/edk2 # ./EfiFsPkg/set_grub_cpu.sh X64
'/usr/src/efifs/grub/include/grub/cpu' -> '/usr/src/efifs/grub/include/grub/x86_64'
root@vm-debian9 /usr/src/edk2 # build -a X64 -b RELEASE -t GCC5 -p EfiFsPkg/EfiFsPkg.dsc
Build environment: Linux-4.9.0-8-amd64-x86_64-with-debian-9.6
Build start time: 22:43:31, Nov.25 2018

WORKSPACE        = /usr/src/edk2
ECP_SOURCE       = /usr/src/edk2/EdkCompatibilityPkg
EDK_SOURCE       = /usr/src/edk2/EdkCompatibilityPkg
EFI_SOURCE       = /usr/src/edk2/EdkCompatibilityPkg
EDK_TOOLS_PATH   = /usr/src/edk2/BaseTools
CONF_PATH        = /usr/src/edk2/Conf

Architecture(s)  = X64
Build target     = RELEASE
Toolchain        = GCC5

Active Platform          = /usr/src/edk2/EfiFsPkg/EfiFsPkg.dsc

Processing meta-data ... done!
Building ... /usr/src/edk2/EdkCompatibilityPkg/Foundation/Library/CompilerStub/CompilerStubLib.inf [X64]
Building ... /usr/src/edk2/MdePkg/Library/UefiDriverEntryPoint/UefiDriverEntryPoint.inf [X64]
Building ... /usr/src/edk2/MdePkg/Library/UefiLib/UefiLib.inf [X64]
"gcc" -g -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -ffunction-sections -fdata-sections -include AutoGen.h -fno-common -DSTRING_ARRAY_NAME=CompilerStubStrings -m64 -fno-stack-protector "-DEFIAPI=__attribute__((ms_abi))" -maccumulate-outgoing-args -mno-red-zone -Wno-address -mcmodel=small -fpie -fno-asynchronous-unwind-tables -Wno-address -flto -DUSING_LTO -Os -Wno-unused-but-set-variable -Wno-unused-const-variable -DMDEPKG_NDEBUG -DDISABLE_NEW_DEPRECATED_INTERFACES -c -o /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EdkCompatibilityPkg/Foundation/Library/CompilerStub/CompilerStubLib/OUTPUT/./memcpy.obj -I/usr/src/edk2/EdkCompatibilityPkg/Foundation -I/usr/src/edk2/EdkCompatibilityPkg/Foundation/Framework -I/usr/src/edk2/EdkCompatibilityPkg/Foundation/Efi -I/usr/src/edk2/EdkCompatibilityPkg/Foundation/Include -I/usr/src/edk2/EdkCompatibilityPkg/Foundation/Include/X64 -I/usr/src/edk2/EdkCompatibilityPkg/Foundation/Efi/Include -I/usr/src/edk2/EdkCompatibilityPkg/Foundation/Framework/Include -I/usr/src/edk2/EdkCompatibilityPkg/Foundation/Include/IndustryStandard -I/usr/src/edk2/EdkCompatibilityPkg -I/usr/src/edk2/EdkCompatibilityPkg/Foundation/Core/Dxe -I/usr/src/edk2/EdkCompatibilityPkg/Foundation/Library/Dxe/Include -I/usr/src/edk2/EdkCompatibilityPkg/Foundation/Library/Dxe/Include/X64 -I/usr/src/edk2/EdkCompatibilityPkg/Foundation/Include/Pei -I/usr/src/edk2/EdkCompatibilityPkg/Foundation/Library/Pei/Include -I/usr/src/edk2/EdkCompatibilityPkg/Foundation/Framework/Ppi/CpuIo -I/usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EdkCompatibilityPkg/Foundation/Library/CompilerStub/CompilerStubLib/DEBUG /usr/src/edk2/EdkCompatibilityPkg/Foundation/Library/CompilerStub/memcpy.c
"gcc" -g -fshort-wchar -fno-builtin -fno-strict-aliasing -Wall -Werror -Wno-array-bounds -ffunction-sections -fdata-sections -include AutoGen.h -fno-common -DSTRING_ARRAY_NAME=UefiDriverEntryPointStrings -m64 -fno-stack-protector "-DEFIAPI=__attribute__((ms_abi))" -maccumulate-outgoing-args -mno-red-zone -Wno-address -mcmodel=small -fpie -fno-asynchronous-unwind-tables -Wno-address -flto -DUSING_LTO -Os -Wno-unused-but-set-variable -Wno-unused-const-variable -DMDEPKG_NDEBUG -DDISABLE_NEW_DEPRECATED_INTERFACES -c -o /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/MdePkg/Library/UefiDriverEntryPoint/UefiDriverEntryPoint/OUTPUT/./DriverEntryPoint.obj -I/usr/src/edk2/MdePkg/Library/UefiDriverEntryPoint -I/usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/MdePkg/Library/UefiDriverEntryPoint/UefiDriverEntryPoint/DEBUG -I/usr/src/edk2/MdePkg -I/usr/src/edk2/MdePkg/Include -I/usr/src/edk2/MdePkg/Include/X64 /usr/src/edk2/MdePkg/Library/UefiDriverEntryPoint/DriverEntryPoint.c

(...)

Scripts/GccBase.lds -Wno-error
"objcopy"  /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Bfs/DEBUG/bfs.dll
cp -f /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Bfs/DEBUG/bfs.dll /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Bfs/DEBUG/bfs.debug
objcopy --strip-unneeded -R .eh_frame /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Bfs/DEBUG/bfs.dll
objcopy  /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Bfs/DEBUG/bfs.dll
cp -f /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Bfs/DEBUG/bfs.debug /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/bfs.debug
"GenFw" -e UEFI_DRIVER -o /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Bfs/DEBUG/bfs.efi /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Bfs/DEBUG/bfs.dll
cp -f /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Bfs/DEBUG/bfs.efi /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Bfs/OUTPUT
cp -f /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Bfs/DEBUG/bfs.efi /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/bfs.efi
[... cut ...]
cp -f /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Btrfs/DEBUG/btrfs.efi /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Btrfs/OUTPUT
cp -f /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Btrfs/DEBUG/btrfs.efi /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/btrfs.efi
cp -f /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Btrfs/DEBUG/*.map /usr/src/edk2/Build/EfiFs/RELEASE_GCC5/X64/EfiFsPkg/EfiFsPkg/Btrfs/OUTPUT

- Done -
Build end time: 22:43:54, Nov.25 2018
Build total time: 00:00:22
```

You may want to refrain from adding any option that I am not adding on my scripts

ElvishJerricco, 2018-11-25:
@pbatard Copy / pasting those commands, I still get the error. Could this be an issue with my GCC version? What version are you using?

```
vagrant@stretch:~$ gcc --version
gcc (Debian 6.3.0-18+deb9u1) 6.3.0 20170516
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

pbatard, 2018-11-25:
```
 # gcc --version
gcc (Debian 6.3.0-18+deb9u1) 6.3.0 20170516
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

ElvishJerricco, 2018-11-25:
@pbatard Huh. Any ideas what other variables there are that could be causing this difference?

- EfiFs rev: d19363a5ff7e0d3ad3ba23c1b7646c1974ac6bec
- EDK II rev: a2481def5da4287570d1a3ed7fe63f737aaf89e8
- Submodules up to date

ElvishJerricco, 2018-11-25:
@pbatard I cannot find that edk2 commit on GitHub, either the main repo or your fork.

pbatard, 2018-11-25:
My bad, was working on an edk2 branch related to pbatard/RaspberryPiPkg where I had applied some EDK2 patches. But I'm still not replicating your issue when using edk2 @ `a2481def5da4287570d1a3ed7fe63f737aaf89e8`:
```
# git show
commit a2481def5da4287570d1a3ed7fe63f737aaf89e8
Author: Vijayenthiran Subramaniam <vijayenthiran.subramaniam@arm.com>
Date:   Wed Nov 21 23:35:59 2018 +0800

    NetworkPkg/IScsiDxe: add debug logs for failed SetVariable attempts

    Add debug messages for failed attempts to write to a variable.

    Cc: Siyuan Fu <siyuan.fu@intel.com>
    Cc: Jiaxin Wu <jiaxin.wu@intel.com>
    Contributed-under: TianoCore Contribution Agreement 1.1
    Signed-off-by: Vijayenthiran Subramaniam <vijayenthiran.subramaniam@arm.com>
    Reviewed-by: Siyuan Fu <siyuan.fu@intel.com>
```

ElvishJerricco, 2018-11-25:
@pbatard What other variables are there? Any other tools that I might need to get a different version of?

ElvishJerricco, 2018-11-25:
Also, any chance your work trees aren't clean?

pbatard, 2018-11-25:
I made sure to issue `git clean -fdx` in edk2, efifs and grub before I rebuilt.

pbatard, 2018-11-25:
Besides, for what is worth, I would encourage you to use the gnu-efi version rather than the EDK2 version of the drivers, as we found that some platforms seemed not to like the EDK2 binaries (pbatard/rufus#1213), whereas they were fine with the gnu-efi ones.

ElvishJerricco, 2018-11-25:
Fair enough. But still, there's some variable we're not accounting for here.

pbatard, 2018-11-25:
Yes, I know. But I'm afraid I don't have the time to investigate this further on my side.