pbatard/EfiFs #12: Image 'hfs_x64.efi' is not an image.

ElvishJerricco, 2018-11-26:
Running OVMF in qemu, going to the UEFI Shell, and entering `load hfs_x64.efi` yields `Image 'hfs_x64.efi' is not an image.` I'm not sure whether it's OVMF or EfiFs at fault, but the other EfiFs drivers that I've tried have worked so I'm assuming it's EfiFs. Both OVMF and EfiFs were built with both their latest releases and their latest commits on `master`.

Are there tools I can use to try and debug this?

pbatard, 2018-11-26:
It's your OVMF. Here's what I get with the last official OVMF binary from https://sourceforge.net/projects/edk2/files/OVMF/ against a gnu-efi Debian 9.6 `hfs_x64.efi`:

[image]

closed 2018-11-26 by pbatard

ElvishJerricco, 2018-11-28:
@pbatard Not sure what I did, but it's loading now. FWIW, the README on that site says:

```
The Latest releases are now hosted on Github: https://github.com/tianocore
```

Though I can't find binary releases there

pbatard, 2018-11-28:
Yes, the source had moved, but the last official OVMF binaries are still on SF, and, where possible, I prefer to validate against official binaries as this reduces the headaches (which you have experienced) of introducing one's own issues through one's build process. With EfiFs being a side project for which I can devote only __very limited time__ compared to my other projects, I have exactly zero incentive to figure out the many ways in which things can break from trying to use the very latest of everything.