pbatard/EfiFs #17: Exception thrown with v1.4

elFarto, 2020-03-04:
I've been testing out v1.4 with the various versions of the EFI shell I've accumulated, and while two of them (the newest versions) work perfectly, one of them manages to cause a machine fault. I'm not entirely sure this is the drivers fault, as I would expect that on all the versions, but it may be the case some other field hasn't been initialised just right.

It's claiming it's version edk2-stable201905, and it's one that came with the OVMF firmware that's on Fedora, not one of the versions I've downloaded directly from their github page. Since it works with one of the other shells, it's not an urgent issue at all.

`UEFI Interactive Shell v2.2
EDK II
UEFI v2.70 (EDK II, 0x00010000)
Mapping table
      FS0: Alias(s):HD0b:;BLK1:
          PciRoot(0x0)/Pci(0x9,0x0)/HD(1,GPT,95EE65AB-C835-4836-85F7-96EC40FE1AFB,0x100800,0x18FF000)
      FS1: Alias(s):HD0c:;BLK2:
          PciRoot(0x0)/Pci(0x9,0x0)/HD(2,GPT,5A066E88-433A-4141-AC41-5DB43836F215,0x800,0x100000)
     BLK0: Alias(s):
          PciRoot(0x0)/Pci(0x9,0x0)

Press ESC in 4 seconds to skip startup.nsh or any other key to continue.
Shell> fs0:
OpenVolume
GetInfo(BECCDD18|'/', 0) <DIR>
Get regular file information
GetInfo(BECCDD18|'/', 600) <DIR>
Get regular file information
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\")
  Reopening <ROOT>
  RET: BECCDD18
Close(BECCDD18|'/') <ROOT>
Close(BECCDD18|'/') <ROOT>
FS0:\> ls
OpenVolume
Open(BECCDD18 <ROOT>, ".")
  Reopening <ROOT>
  RET: BECCDD18
GetInfo(BECCDD18|'/', 0) <DIR>
Get regular file information
GetInfo(BECCDD18|'/', 600) <DIR>
Get regular file information
Close(BECCDD18|'/') <ROOT>
GetInfo(BECCDD18|'/', 0) <DIR>
Get regular file information
GetInfo(BECCDD18|'/', 600) <DIR>
Get regular file information
SetPosition(BECCDD18|'/', 0) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
error: not a regular file.
Unable to obtain the size of 'bin': [17]
Read(BECCDD18|'/', 602) <DIR>
error: not a regular file.
Unable to obtain the size of 'sbin': [17]
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Read(BECCDD18|'/', 602) <DIR>
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\lost+found")
  RET: BEB2F998
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\.viminfo")
  RET: BEB31018
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\.ash_history")
  RET: BEB5F298
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\usr")
  RET: BEB5F518
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\sys")
  RET: BEB5F018
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\etc")
  RET: BEB60598
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\lib")
  RET: BEB60D18
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\proc")
  RET: BEB60018
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\dev")
  RET: BEB61498
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\boot")
  RET: BEB61E18
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\run")
  RET: BEB61018
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\pkgs")
  RET: BEC57E98
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\bin")
error: not a regular file.
Could not open file '\bin': [17]
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\sbin")
error: not a regular file.
Could not open file '\sbin': [17]
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\lib64")
  RET: BEB2EC18
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\var")
  RET: BEB2F598
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\root")
  RET: BEB2ED18
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\.bash_history")
  RET: BEB2E718
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\tmp")
  RET: BEB2DB98
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\system")
  RET: BEB2D918
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\mnt")
  RET: BEB2D298
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\home")
  RET: BEB2D518
Close(BECCDD18|'/') <ROOT>
OpenVolume
Open(BECCDD18 <ROOT>, "\srv")
  RET: BEB2D618
Close(BECCDD18|'/') <ROOT>
!!!! X64 Exception Type - 03(#BP - Breakpoint)  CPU Apic ID - 00000000 !!!!
RIP  - 00000000BE05CEB7, CS  - 0000000000000038, RFLAGS - 0000000000000246
RAX  - 0000000000000000, RCX - 0000000000000000, RDX - 0000000000000402
RBX  - 00000000BFF6FC70, RSP - 00000000BFF6FC40, RBP - 00000000000007FF
RSI  - 00000000000007FF, RDI - 00000000BEB31E30
R8   - 0000000000000001, R9  - 000000000000009F, R10 - 0000000000000000
R11  - 0000000000000000, R12 - 00000000BEC58E18, R13 - 00000000BFF6FF40
R14  - 0000000000000000, R15 - 00000000BF09F818
DS   - 0000000000000030, ES  - 0000000000000030, FS  - 0000000000000030
GS   - 0000000000000030, SS  - 0000000000000030
CR0  - 0000000080010033, CR2 - 0000000000000000, CR3 - 00000000BFC01000
CR4  - 0000000000000668, CR8 - 0000000000000000
DR0  - 0000000000000000, DR1 - 0000000000000000, DR2 - 0000000000000000
DR3  - 0000000000000000, DR6 - 00000000FFFF0FF0, DR7 - 0000000000000400
GDTR - 00000000BFBEE698 0000000000000047, LDTR - 0000000000000000
IDTR - 00000000BF5AF018 0000000000000FFF,   TR - 0000000000000000
FXSAVE_STATE - 00000000BFF6F8A0
!!!! Find image based on IP(0xBE05CEB7) /builddir/build/BUILD/edk2-edk2-stable201905/Build/OvmfX64/DEBUG_GCC5/X64/ShellPkg/Application/Shell/Shell/DEBUG/Shell.dll (ImageBase=00000000BDFFA000, EntryPoint=00000000BE051258) !!!!`

pbatard, 2020-03-04:
Thanks for the report. From the content above, I'm going to assume that you are using the `ext` driver, right?

I'm also seeing quite a few `error: not a regular file.` being thrown in your listing. Can you confirm that these are symlinks? What happens if you try to list the content of a directory that doesn't have symlinks?

Alternatively, have you been able to replicate this issue with other file system drivers than `ext`?

I guess one thing I'd like you to do, if you can, is upload the OVMF firmware you use somewhere, that produces the error, so that I can test it with QEMU and see if I can reproduce the problem. Or you can compress it and e-mail it to pete@akoe.ie.

It may also help if you indicate how you run your test. What virtual platform are you using OVMF against, and what is the command line you use to launch it? Maybe also, if you can mount a virtual disk, can I ask you to try to test against the `.img` you'll find inside https://efi.akeo.ie/test/ext2.zip, since that is the disk target I'm using for my testing. Knowing that whether it also fails for you when trying to list its content would remove one question mark on whether the content is relevant, as well as, if it does fail too, give me a better chance to reproduce the issue.

pbatard, 2020-03-04:
For the record I just ran a quick test with symlinks and whereas I got the `error: not a regular file.` message, everything seemed fine. Of course, for the time being, I'm using a custom a very recent version of OVMF, that I built, so I didn't really expect that test to fail.

elFarto, 2020-03-05:
Yes, it's the ext2 driver, on an ext4 filesystem.
/bin and /sbin are symlinks, but they don't seem to be the cause of the issue. Running ls in an empty directory works fine, but creating an empty file in there will cause it to crash.
Doing an ls on your ext2 image in the root directory has the same issue.
I've uploaded my OVMF copy [here](https://github.com/elFarto/efi). I'm running it under qemu/kvm.

pbatard, 2020-03-05:
Thanks. I was able to replicate your issue with your OVMF in qemu.

The first thing I will point out then is that it's not a regression from 1.4, as the same problem occurs with the 1.3 version of the driver. And I also seem to get the crash regardless of the file system being used.

Considering that this happens after we close all access to the drive, and that I'm not crawling under _"Your file system drivers crash on my system!"_ reports, I'm very inclined to think that this was a transient bug in EDK2, that happened to make it into an OVMF release, but that has long since been fixed, so I'm not sure I wanna spend too much time investigating it. Real hardware users don't seem to be affected by it, and, unless proven otherwise, virtual users should easily be able to fix that issue, if they happen to run into it, by upgrading or downgrading to a non affected version of OVMF.

As such, unless someone can demonstrate that this is something that didn't happen with some older version of EfiFs, or that such a crash may still be experienced with current OVMF, I think I will just close this issue.

closed 2020-03-05 by pbatard

elFarto, 2020-03-05:
I agree, I just wanted to make sure it wasn't a regression. Thanks for spending time on it anyway.