pbatard/EfiFs #22: NTFS build

startergo, 2020-05-30:
Building in VS2019 errors:
```
Severity	Code	Description	Project	File	Line	Suppression State
Error	NETSDK1013	The TargetFramework value '' was not recognized. It may be misspelled. If not, then the TargetFrameworkIdentifier and/or TargetFrameworkVersion properties must be specified explicitly.	ntfs	C:\Program Files\dotnet\sdk\3.1.300\Sdks\Microsoft.NET.Sdk\targets\Microsoft.NET.TargetFrameworkInference.targets	93	

```

pbatard, 2020-05-30:
Not seeing any issue with a __vanilla__ up to date install of VS2019.

For the record, VS2019 is what I am using for EfiFs development, so I am checking that VS2019 does work on regular basis.

Most likely, you didn't change the default target VS2019 is compiling to (which would be `aa64`) and you failed to follow the README instructions, that tell you exactly what you need to do to install ARM/ARM64 support. So you either need to install support for those or change the compilation target in VS2019.

startergo, 2020-05-30:
[image]
 I don't care about ARM. Only x64.
I built everything in Debian, but VS2019 comes with a lot of syntax errors for some reason.

pbatard, 2020-05-30:
Again, I am not seeing that error in a vanilla installation of VS2019, so there's not much I can do to help. My guess is that you didn't properly install the C++ compilation environment, so you will have to sort that out, as I have never seen the error you report.

startergo, 2020-06-17:
I am now able to compile through CMD. It was related to 
https://github.com/tianocore/edk2-pytool-extensions/issues/207#issuecomment-645466243
Still having troubles with the Visual Studio F5 though.

pbatard, 2020-06-19:
Okay. I'm going to close this issue since the problem is with your environment and there isn't much help I can provide.

closed 2020-06-19 by pbatard