pbatard/EfiFs #23: refind will not start with f2fs_x64.efi 1.6

glenn4uk, 2020-06-19:
Using refind 0.12.0-1 to boot debian bullseye.  Refind does not even start when using the 1.6 version of f2fs efi driver.  No problem with 1.5.

pbatard, 2020-06-19:
Please run everything from the UEFI Shell and report what you see after you running the command `set FS_LOGGING 4`.

glenn4uk, 2020-06-19:
Sorry - I have a bit of a catch-22 here.  My UEFI firmware does not include a shell.  I have to load a shell externally from refind but refind won't even start with the f2fs 1.6 driver in its driver search path.

glenn4uk, 2020-06-19:
Sorry - I have a bit of a catch-22 here.  My UEFI firmware does not include a shell.  I have to load the shell externally from refind but refind won't even start with the f2fs 1.6 driver in its driver search path.

marlock9, 2020-11-14:
Same with xfs version 1.6. Just stucks after loading the driver.
Pictures of efi shell after set FS_LOGGING 4:
[image]
1.5 driver: [image]

pbatard, 2020-11-17:
@marlock9, thanks for the report. 

My guess is that you are hitting one of the new `ASSERT`s that were introduced as part of c67e04a5089e7542453e8f3fd849ebf947d5e4fb, as by the looks of it, the 1.6 driver succeeds in enabling the file system for your XFS volumes, but trips when processing the subsequent non XFS devices you have on your system. Most likely, your firmware is feeding something like a NULL Device Path or something, and this is something we didn't `ASSERT` on before. (NB: The `FSInstall: D` difference is a benign issue that I have now fixed in 6b730d260f2487557fdc7fd8253be514e8a6757f).

Can I therefore please ask you to run the same test with this [`xfs_test1.efi`](https://rufus-web.akeo.ie/testing/xfs_test1.efi)  x64 driver and report what you get so that I can validate the location of the issue?

pbatard, 2020-11-18:
Nevermind, I'm pretty sure I have identified the issue now. This is due to version 1.6 [freeing the Device Path](https://github.com/pbatard/efifs/commit/c67e04a5089e7542453e8f3fd849ebf947d5e4fb#diff-8f15c6da308c1cd5c94ff880ad21bea303c41779315b8086c75b7fe41b81de58R77-R78), which we should of not do, as this is a system allocated resource.

So, of course, once you start freeing Device Paths for volumes that the system might still be trying to access, all kind of bad things happen...

I've actually managed to replicate the error above on one of my systems, and confirmed that not trying to free the Device Path fixed it. I'll produce a 1.7 release of EfiFs with this fix when I get a chance... In the meantime, you may want to test with this [`xfs_test2.efi`](https://rufus-web.akeo.ie/testing/xfs_test2.efi) x64 driver, that includes the fix, to confirm that it now works as expected.

pbatard, 2020-11-18:
I have now released EfiFs v1.7 that should fix the issue. If there is still a problem, let me know.

commit 959c6e50 (2020-11-18): don't free Device Paths

* These are system resources that shouldn't be freed by drivers.
* Closes #23
closed 2020-11-18 by pbatard