pbatard/EfiFs #25: NTFS x64 driver not working in Workstation Pro Shell?

aturnbul, 2020-08-12:
This may be an inappropriate place to ask, and please excuse me, if so. But it may be related the the driver code, not just a particular shell, so here goes.

I have the latest versions of your wonderful library of drivers and have compiled them using both VS2019 Enterprise and EDK2 (2020-08-10 chain) targeting <arch>=x64. Everything works well in QEMU with OVMF but in VMWare Workstation Pro's shell (version 2.31 - the latest), although loading the ntfs_x64.efi driver returns success, most commands that query ANY disk i/o (not just ntfs or the device it's on) cause the shell to hang. Certainly map -r or reconnect -r cause a hang.

I also noticed a similar issue with QEMU and OVMF when the virtual disk image was created with various tools or configurations. Things as simple as creating and initializing a raw image in Disk Management as GPT, then partitioning it with a protective partition, an ESP (FAT32) partition, and an NTFS partition. QEMU boots the shell but hangs after loading the ntfs_x64.efi driver and trying to connect the 3rd (NTFS) partition. The ESP partition was fine.

Thanks for the project and thanks for any help you might suggest.

pbatard, 2020-08-12:
For extensive logging output, set the `FS_LOGGING` shell variable to `4`. Then you should see where it hangs and you can have a closer look at the source. What's the output you get with `FS_LOGGING` set to `4`?

aturnbul, 2020-08-12:
Thank you for your very quick response! The issue appears to be that the filesystem is not recognized as ntfs. See attached for the specifics, but it's not much more that that.

[NTFS Filesystem Error.pdf](https://github.com/pbatard/efifs/files/5065295/NTFS.Filesystem.Error.pdf)

The file itself was created by Disk Management Version: 10.0.19041.329 (Windows 10 Enterprise, v2004) as a vhd, initialized as a GPT disk then formatted with Diskpart (also W10 E) which created a a 15M Reserved partition (protective) and a 132M Primary partition that was then formatted as NTFS (format fs=ntfs quick). The resulting vhd mounts perfectly under Windows.

Results of Qemu-img:
D:\VCsp_Projects\UEFI\efifs>qemu-img info "D:\VMWare VMs\NTFS Partition on GPT Disk.vhd"
image: D:\VMWare VMs\NTFS Partition on GPT Disk.vhd
file format: raw
virtual size: 150 MiB (157286912 bytes)
disk size: 150 MiB

Any thoughts? The driver doesn't seem to like Reserved partitions or, possibly, any partitions before the NTFS partition.

Thanks for any help.

pbatard, 2020-08-12:
Well, you're going to have to intrument [this](https://github.com/pbatard/efifs/blob/master/src/grub_file.c#L442-L448) which means instrumenting `grub_ntfs_dir()` and `grub_ntfs_mount()` from the [GRUB source](https://git.savannah.gnu.org/cgit/grub.git/tree/grub-core/fs/ntfs.c), because the error you get ultimately [comes from there](https://git.savannah.gnu.org/cgit/grub.git/tree/grub-core/fs/ntfs.c#n966)...

I do understand that you would probably like me to be the one investigating this issue, but I'm afraid I have other priorities at the moment, and I genuinely don't expect to be in a position to do that for 2 or 3 months... and probably much longer (because, as long as this is not an issue that doesn't affect my [UEFI:NTFS bootloader](https://git.savannah.gnu.org/cgit/grub.git/tree/grub-core/fs/ntfs.c#n966), anything EfiFs related is super low priority).

So, considering that you are both in a position to recompile the drivers and test the issue, I hope that you are okay with investigating it yourself (by adding extra logging statements until you can pinpoint what's really happening). Once you have done that, I'll be happy to address whatever issue you found.

aturnbul, 2020-08-13:
Fair enough and I certainly understand. I'll work on this as time allows and let you know what I find.

pbatard, 2021-01-22:
I'm going to assert that this is a duplicate of #23 that was fixed in 1.7. &rarr; Closing.

closed 2021-01-22 by pbatard