pbatard/EfiFs #29: Build errors

dbermond, 2021-03-18:
I'm getting errors when building for `x64`, `ia32` and `arm` with gcc.

For`x64`:
```
[LD] btrfs_x64.efi
/usr/lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/ld: /storage/linux/src/efifs/src/..//grub/grub-core/fs/btrfs.o:btrfs.c:(.text+0x1839): undefined reference to `grub_raid6_recover_gen'
collect2: error: ld returned 1 exit status
make[2]: *** [/storage/linux/src/efifs/src/../Make.common:166: btrfs_x64.efi] Error 1
make[2]: Leaving directory '/storage/linux/src/efifs/src'
make[1]: *** [Makefile:18: btrfs] Error 2
make[1]: Leaving directory '/storage/linux/src/efifs/src'
make: *** [Makefile:16: all] Error 2
```

For `ia32`:
```
[LD] btrfs_ia32.efi
/usr/lib/gcc/i686-w64-mingw32/10.2.0/../../../../i686-w64-mingw32/bin/ld: /storage/linux/src/efifs/src/..//grub/grub-core/fs/btrfs.o:btrfs.c:(.text+0x178e): undefined reference to `grub_raid6_recover_gen'
collect2: error: ld returned 1 exit status
make[2]: *** [/storage/linux/src/efifs/src/../Make.common:166: btrfs_ia32.efi] Error 1
make[2]: Leaving directory '/storage/linux/src/efifs/src'
make[1]: *** [Makefile:18: btrfs] Error 2
make[1]: Leaving directory '/storage/linux/src/efifs/src'
make: *** [Makefile:16: all] Error 2
```

For `arm`:
```
arm-none-eabi-gcc -I/storage/linux/src/efifs/gnu-efi//gnuefi -I/storage/linux/src/efifs/gnu-efi//gnuefi/../inc -I/storage/linux/src/efifs/gnu-efi//gnuefi/../inc/arm -I/storage/linux/src/efifs/gnu-efi//gnuefi/../inc/protocol -Wno-error=pragmas -marm -fpic  -g -O2 -Wall -Wextra -Werror -fshort-wchar -fno-strict-aliasing -ffreestanding -fno-stack-protector -fno-stack-check -fno-stack-check -fno-merge-all-constants -DCONFIG_arm -c /storage/linux/src/efifs/gnu-efi//gnuefi/reloc_arm.c -o reloc_arm.o
/storage/linux/src/efifs/gnu-efi//gnuefi/reloc_arm.c:40:10: fatal error: elf.h: No such file or directory
   40 | #include <elf.h>
      |          ^~~~~~~
compilation terminated.
make[2]: *** [/storage/linux/src/efifs/gnu-efi//gnuefi/../Make.rules:52: reloc_arm.o] Error 1
make[2]: Leaving directory '/storage/linux/src/efifs/gnu-efi/arm/gnuefi'
make[1]: *** [Makefile:80: gnuefi] Error 2
make[1]: Leaving directory '/storage/linux/src/efifs/gnu-efi'
make: *** [Make.common:158: gnu-efi/arm/lib/libefi.a] Error 2
```

It builds fine for `aa64`.

Commands used to build:
```
$ git clone --recursive https://github.com/pbatard/efifs.git
$ cd efifs
$ patch -d grub -Np1 -i ../0001-GRUB-fixes.patch
$ unset -v CFLAGS
$ make -j1 ARCH='x64'
$ # (or make -j1 ARCH='ia32')
$ # (or make -j1 ARCH='arm' CROSS_COMPILE='arm-none-eabi-')
```

System information:
- OS: Arch Linux x86_64
- mingw-w64-gcc: 10.2.0
- arm-none-eabi-gcc: 10.2.0

dbermond, 2021-03-21:
For both `x64` and `ia32`, a git bisect shows that the first bad commit is 8d005299c5ea0a1e924d659b0bc88af1483d275a. Bisected between v1.3 and v1.7, since I found that v1.3 builds fine for me.

I could solve the issue for `arm` by installing a system package that complements the cross compiler. It was a only a missing header.

pbatard, 2021-08-08:
Sorry for the delay. The issues should be fixed now (except for the missing `elf.h` header which has to do with the toolchain/system and not EfiFs).

closed 2021-08-08 by pbatard

dbermond, 2021-08-10:
Thank you for the reply.

I can confirm that this is now fixed. Thank you for fixing this.

Offtopic: it fails to build with more than one make job. For example, it fails with `make -j8` and builds fine with `make -j1`. Is this intentional?

pbatard, 2021-08-10:
Not intentional, but it's a case of "I really can't be bothered enough to try to fix this". As you can see from the time it took to look into the issue you reported, EfiFs is a low priority project for me (and for good reason: it just doesn't count enough users compared to other projects I'm involved with), so it just doesn't make sense to sink a lot of time to get minor issues like this one ironed out.

Instead, I have now added [GitHub Actions workflows for all the major build paths](https://github.com/pbatard/efifs/tree/master/.github/workflows), so that I'll be able to instantly detect breakage like the one you originally reported. And considering that the `make` invocation there is not a multijob one, as long as I am not seeing any breakage from GitHub actions, I just can't be bothered to look into it (but I will accept a fix if someone is willing to send one).

dbermond, 2021-08-10:
Got it. Thank you for the reply again :)

commit 5df60e20 (2021-08-08): fix compilation issues

* Closes #29
* Also update GRUB and gnu-efi to latest