pbatard/EfiFs #33: xfs error in StartImage

Kruptein, 2021-08-31:
I wanted to try out xfs with refind and downloaded the latest xfs_x64.efi file from the releases to my /efi/EFI/refind/drivers_x64 folder, but I'm not able to boot into the linux image.

Loading up an efi shell and going to that folder on fs0 and trying to `load xfs_x64.efi` I get one line: `Image FS0:\EFI\refind\drivers_x64\xfs_x64.efi error in StartImage: Load Error`.

The sha256 matches the one in the gh action, so it doesn't seem to be a corrupt file, any idea what I'm doing wrong ?

pbatard, 2021-08-31:
I just tested `xfs_x64.efi` on two separate x86-64 UEFI systems (one fairly old and one fairly new), and saw no error when issuing `load xfs_x64.efi`.

Typically, I'd expect `Load Error` if you tried to load a binary that doesn't match the architecture of your UEFI firmware.

For instance, you may need to use the IA32 version of the drivers even if you have an x86 64-bit CPU if your UEFI firmware happens to be for 32-bit (which can happen on some platforms where the manufacturer is eager to save space).

At any rate, it looks like your issue is environmental and not an issue with the `xfs_x64.efi` driver.

Kruptein, 2021-08-31:
Ok thanks, I'll investigate it a bit more on my end then, it's a new dell precision laptop so I expected it to just be regular x64

Kruptein, 2021-08-31:
ah ok I found some more info for you I think.
What I was doing up to know whas loading the efi shell from refind.

I now loaded the efi shell from my live arch iso and when I first load the driver it shows more things and when I run it a second time it shows the startimage error. So I'm going to assume the latter is irrelevant and the first load info dump is going to give more info.

FS driver installed
FSBindingSupport
FSBindingStart
error: not a XFS filesystem.

it's fs4 that is my xfs partition though, I'm not familiar enough with efi shell to know how to load the driver specifically for that filesystem. If I switch to fs4: I no longer have access to the driver which is located in fs0.

pbatard, 2021-08-31:
Ah, yes, `StartImage: Load Error` is what you're going to get if you attempt to load a driver twice, and I suspect rEFInd already had an XFS driver loaded, so that makes sense.

Not sure why your XFS file system is not being recognized though, because I still don't see any issue when creating a USB drive with one ESP partition including the UEFI Shell and the `xfs_x64.efi` driver and a second XFS partition (created from Debian 11 using an up to date `mkfs.xfs`).

There I was able to issue:
```
load xfs_x64.efi
map -r
fs1:
```
and I validated that I was able to access and read content from the XFS partition (`FS1:`).

All in all, I don't see anything for me to address here. My testing shows that the latest XFS driver appears to work as expected. So I'm afraid you'll have to carry out further troubleshooting on your own...

polarathene, 2023-07-26:
Just for reference, while I haven't tried with upstream GRUB or the released XFS driver on this project, I did recently encounter an issue with [Ventoy (uses GRUB drivers) not being able to detect an XFS partition](https://github.com/ventoy/Ventoy/issues/2494#issuecomment-1646674091).

It's possible this issue was related to my findings there, due to changes introduced to XFS (_since 5.10 kernel released Dec 2020, default via `mkfs.xfs` since `xfsprogs` 5.15 released April 2022_). If an XFS partition was created with `bigtime=1`, the XFS driver from GRUB probably would have had trouble accessing it.

It was addressed in June 2021 with the GRUB 2.06 release, but not entirely. A 2nd follow-up commit arrived in Feb 2023 which will be part of GRUB 2.12. If the issue hasn't been fixed yet with current XFS driver / GRUB, and anyone runs into the problem hopefully that's helpful 👍