pbatard/EfiFs #34: Can't see Ext4 partition (I'm using VirtualBox)

q2dg, 2021-09-22:
Hello.

I've downloaded "ext2_x64.efi" fle from your web but, after having loaded, when I remap filesystem in VirtualBox's UEFI Shell, nothing happens: my Ext4 partition continues just recognized as "blk", not "fs".

I don't know if it is a problem of your driver or it's a VirutualBox UEFI's implementation's one but I think it was worthy to open an issue about this, in order to debug the problem (hope I'm not mistaken).

Thanks a lot for your splendid project.

pbatard, 2021-09-22:
I run regular tests (using QEMU) for the ext4 driver, and I am not aware of any issue. This is a test I just re-ran with the exact `ext2_x64.efi` driver that's published on the web site:

[image]

IIRC, it's not the first time people report issues with VirtualBox, and I'm afraid I have no time (or interest) for debugging what happens there when the driver works fine with QEMU or real systems. You will have to investigate this problem yourself, sorry.

closed 2021-09-22 by pbatard

q2dg, 2021-09-22:
I understand, no problem!
Thanks!