pbatard/EfiFs #37: f2fs driver not working.

arun54321, 2022-02-12:
[image]
[image]

pbatard, 2022-02-12:
Works fine with QEMU (which is how I test the driver):
[image]

This is using the test F2FS image from https://efi.akeo.ie/test/f2fs.zip

I suggest you enter `set FS_LOGGING 4` before loading the f2fs driver to see how your `ls` operation differs from mine:

[image]

I'm afraid that, unless you can provide me with an f2fs image that can replicate the issue with in QEMU, and with details on how exactly you created that image, there's not much I'll be able to do to help considering that, as far as my testing is concerned, the f2fs driver works just fine.

pbatard, 2022-02-12:
Probably won't work, especially as I don't see support for f2f2 encryption or compression in the GRUB driver.

I guess the question becomes: _"Can you actually access that same device from GRUB?"_

Please bear in mind that these drivers are derived directly from the ones found in the GRUB project, which are fairly basic by nature, and are therefore subject to the same limitations, if not more, since, considering that this project is a byproduct of work that I did because I needed a UEFI NTFS driver (and only that driver), I also dropped any additional feature from the GRUB drivers that were causing me compilation headaches.

This means that, if GRUB can't access your device, then EfiFs definitely won't and, even if GRUB can actually access it, if that device using additional features such as encryption and/or compression, chances are that I might have dropped such feature on account that the time I can spend on this project is very limited.

So, really, it's up to you to find whether you can access your device when using GRUB and then, if you actually can, submit a patch to add the extra features, as I'm afraid genuinely don't have the scope to do that work for you.

arun54321, 2022-02-12:
Only the files in /home/user folder are encrypted (fscrypt). they are not required for booting.

pbatard, 2022-02-12:
I don't think it matters, since, if there's any encryption on the file system, it will be advertised in the global feature, which may result in the GRUB driver rejecting it wholesale.

Again, you will need to test whether you can access your device from GRUB. There's nothing more I can do for you here unless you are willing to first check what happens when using GRUB.

arun54321, 2022-02-12:
Just now installed grub for testing. it doesn't work.