pbatard/EfiFs #39: Dose it work on mobile devices?

ChenQian17, 2022-09-02:
I tried to apply f2fs to UEFI on mobile device, but failed.
I found that the program runs to the function `grub_f2fs_check_dentries` and return after looping the function `grub_f2fs_test_bit_le` 27 times. 
I printed value of ctx->bitmap, all zero.
How should I debug it?

pbatard, 2022-09-03:
> How should I debug it?

By continuing what you're doing and investigating why the code doesn't see what you expect.

You should also attempt to boot the latest GRUB and see if it can access your partition (since the EfiFs file system drivers are directly derivated from GRUB, so they have the same limitations as the current GRUB driver)

Please bear in mind that the f2fs driver in GRUB is very recent, so it's possible it may not support all versions of f2fs. It looks to me like if `ctx->bitmap` is all zero, then it means that either you're not accessing the right f2fs partition, or you f2fs partition is created in a way that GRUB does not support, and therefore EfiFs does not support either.

pbatard, 2022-12-21:
Since there has not been any update I will close this issue.

closed 2022-12-21 by pbatard