pbatard/EfiFs #40: Update: Adding a driver to a UEFI firmware

DiMeneses, 2022-10-26:
The guide present in the wiki doesn't seem to work anymore, as I can't find any `.rsrc` folder in `vmware-vmx.exe`.

Is this dependent on any `vmware` version? If so, each one?

Is there any other alternative?

Thanks for your attention.,

pbatard, 2022-10-26:
I'm afraid I no longer use VMWare and I don't have the time to perform any research to update the guide. Like most documentation, this is time dependent of the software that existed at the time it was written.

Mostly the VMWare UEFI firmware was used as an __example__ of how one can inject an NTFS driver, but the goal is not to be a comprehensive guide that will apply to everything. As such, I'm afraid you will have to do your own research if you want to apply it to different software than the one that was used for this example.

closed 2022-10-26 by pbatard

DiMeneses, 2022-10-26:
No problem, it also served as a heads up. 

If I come up with an alternative solution I'll post it here in case you want to update the guide with it.

DiMeneses, 2022-11-04:
### Proposed update
- Package your `.efi` driver as per the `efifs` instructions: "Creating the UEFI firmware module".
- With UEFITool, press `ctrl+O`  to open an `OVMF.fd` (open virtual machine firmware) image. 
- You can find the latest build of OVMF  [here](https://www.kraxel.org/repos/jenkins/edk2/), under the name `edk2.git-ovmf-x64-0`.
- Use `ctrl+F` to `GUID search` for `961578FE-B6B7-44C3-AF35-6BC705CD2B1F`.
- Double click the `"GUID pattern"` message at the bottom.  
- Right-click to `insert after` the module of your choice. 
- Choose your `.ffs` file. 
- Press `ctrl+S` to save the image. Choose a new name such as `yourOVMF.fd` or overrride the previous image. 
- Test it with `QEMU`:
```sh
qemu-system-x86_64 -bios yourOVMF.fd -net none
```