pbatard/EfiFs #42: Next update

janos666, 2023-10-02:
I see you usually update these drivers roughly once a year.
Could you please include Btrfs "extent tree v2" support in this year's update? It presumably increases the mount time significantly.
I love how my box boots directly from Btrfs after injecting your driver into the motherboard's UEFI firmware but boot time is ridiculously slow this way (I guess the CPU is operating at it's lowest possible frequency during that time and these drivers might not be optimized for speed either).
The kernel mounts the HDD-based RAID filesystem without a few seconds with or without that new Btrfs feature but the motherboard firmware takes a few minutes in comparison. So, I am not sure how much of a speedup I should expect if you include that support in your driver but given the scale, a (let's guess) ~50% reduction in firmware-mount time would be very perceivable (could shave off an entire minute or so).

janos666, 2023-10-02:
Thank you for the information (and your past work on this regardless).

pbatard, 2023-11-08:
By the way, since it appears you are also under the impression that I get to decide the features that get added to the file system code, and can therefore pick things at at will, please be aware that this is very far from the truth.

For a file system feature to be proposed by EfiFs, it __must__ first be added to the [GRUB project](https://www.gnu.org/software/grub/), since I do not perform any development when it comes to the file systems at all (that would be way too much work!), but am merely reusing the existing GRUB code, and I am [not seeing anything related to Btrfs "extent tree v2" support in there](https://git.savannah.gnu.org/cgit/grub.git/log/).

So, if you do want that feature in EfiFs, you're going to first have to wait until someone thinks that this is a feature that GRUB needs, and is willing to implement it there or you will have to propose a patch to GRUB yourself because that is the only way you will see a specific file system functionality appear in EfiFs.

As a result, I will close this issue.

closed 2023-11-08 by pbatard