pbatard/EfiFs #44: rEFInd can't find linux kernel on /boot with XFS filesystem when "crc=1"

Sunshine-Planet, 2024-03-02:
## env
arch => x86_64
/efi => ESP partion
root  partition => XFS
mkfs.xfs version 6.5.0 in archlinux livecd 2024.02.01

## question
I format root partition with command ```mkfs.xfs /dev/sda2```
install refind and put efifs diver ```xfs_x64.efi``` in /efi/EFI/refind/drivers_x86
but after boot into refind menu, it can't autoly show linux kernel boot option

I tried to set off ```mkfs.xfs -m bigtime=0 /dev/sda2``` mentioned in #33 but it doesn't work
then set off ```-m inobtcount=0 -m finobt=0 -m rmapbt=0``` and also doesn't work

I set off crc with ```mkfs.xfs -m crc=0 /dev/sda2``` and finally linux kernel shows on refind, boot into linux.

In mkfs.xfs man page
> Formatting a filesystem without CRCs selects the V4 format, which is deprecated and will be removed from upstream in September 2030. Distributors may choose to withdraw support for the V4 format earlier than this date. Several other options, noted below, are only tunable on V4 formats, and will be removed along with the V4 format itself.

Is ```xfs_x64.efi``` doesn't support xfs v5 with crc=1 which mkfs.xfs default parameter yet?

pbatard, 2024-03-02:
EfiFs will get updated to support xfs v5 but we have to wait on GRUB (which is the parent project where all the EfiFs file systems derive from) to actually get updated to support xfs v5.

From what I can see, this only happened in November last year in https://git.savannah.gnu.org/cgit/grub.git/commit/?id=aa7c1322671eef48ba72d3b733da37e63eb37328 so, obviously, and especially as EfiFs releases are quite far and in-between (since I'm afraid I can't prioritize EfiFs over other more important projects of mine), EfiFs does not support that feature yet.

It's unfortunate that some distros pushed for using v5 early without giving enough time for downstream elements to actually be updated to support v5 (considering that GRUB 2.12 has only been released earlier this year, I would have give at least one year of time for people to upgrade from GRUB 2.06, which is still __widely__ used and does not support xfs v5 before making v5 default in `mkfs.xfs`).

So, all I can say is, as long as GRUB does get updated to support XFS v5 features, then EfiFs will ultimately get updated to support the same features. But I cannot tell you when the next release of EfiFs is going to be (it's probably at least 4 to 6 months away, at best, at this stage) and if you use EfiFs, I would advise you to take into consideration that, when properties of a file system that did were not supported by GRUB until a few months ago get enabled by default in Linux, you are indeed going to have to do your own homework and disable them to work with EfiFs, until there is a new release.

Sunshine-Planet, 2024-03-02:
Thank you and your detailed explanation, I will conside more about a new file system properties supported status.

closed 2024-03-02 by Sunshine-Planet

wahooli, 2025-07-30:
Ventoy, which uses grub apparently has recently fixed this https://github.com/ventoy/Ventoy/pull/3284

drolevar, 2026-03-13:
@pbatard It's fixed in the GRUB 2.14 release from January. I've updated the submodule, fixed the discrepancies, and it works on my system. Please review PR #50. Thanks.