pbatard/EfiFs #47: Build failure for the `arm` architecture

dbermond, 2024-06-06:
I'm getting the following build error in version 1.10 for the `arm` architecture:

```
  CC       entropy_common.o
In file included from grub/grub-core/lib/zstd/fse.h:303,
                 from grub/grub-core/lib/zstd/entropy_common.c:41:
grub/grub-core/lib/zstd/bitstream.h:395:19: error: conflicting types for 'BIT_readBits'; have 'grub_size_t(BIT_DStream_t *, U32)' {aka 'unsigned int(BIT_DStream_t *, long unsigned int)'}
  395 | MEM_STATIC size_t BIT_readBits(BIT_DStream_t* bitD, U32 nbBits)
      |                   ^~~~~~~~~~~~
grub/grub-core/lib/zstd/bitstream.h:123:21: note: previous declaration of 'BIT_readBits' with type 'grub_size_t(BIT_DStream_t *, unsigned int)' {aka 'unsigned int(BIT_DStream_t *, unsigned int)'}
  123 | MEM_STATIC size_t   BIT_readBits(BIT_DStream_t* bitD, unsigned nbBits);
      |                     ^~~~~~~~~~~~
grub/grub-core/lib/zstd/bitstream.h:404:19: error: conflicting types for 'BIT_readBitsFast'; have 'grub_size_t(BIT_DStream_t *, U32)' {aka 'unsigned int(BIT_DStream_t *, long unsigned int)'}
  404 | MEM_STATIC size_t BIT_readBitsFast(BIT_DStream_t* bitD, U32 nbBits)
      |                   ^~~~~~~~~~~~~~~~
grub/grub-core/lib/zstd/bitstream.h:148:19: note: previous declaration of 'BIT_readBitsFast' with type 'grub_size_t(BIT_DStream_t *, unsigned int)' {aka 'unsigned int(BIT_DStream_t *, unsigned int)'}
  148 | MEM_STATIC size_t BIT_readBitsFast(BIT_DStream_t* bitD, unsigned nbBits);
      |                   ^~~~~~~~~~~~~~~~
grub/grub-core/lib/zstd/fse.h:534:17: error: conflicting types for 'FSE_encodeSymbol'; have 'void(BIT_CStream_t *, FSE_CState_t *, U32)' {aka 'void(BIT_CStream_t *, FSE_CState_t *, long unsigned int)'}
  534 | MEM_STATIC void FSE_encodeSymbol(BIT_CStream_t* bitC, FSE_CState_t* statePtr, U32 symbol)
      |                 ^~~~~~~~~~~~~~~~
grub/grub-core/lib/zstd/fse.h:380:13: note: previous declaration of 'FSE_encodeSymbol' with type 'void(BIT_CStream_t *, FSE_CState_t *, unsigned int)'
  380 | static void FSE_encodeSymbol(BIT_CStream_t* bitC, FSE_CState_t* CStatePtr, unsigned symbol);
      |             ^~~~~~~~~~~~~~~~
make: *** [Make.common:202: grub/grub-core/lib/zstd/entropy_common.o] Error 1
make: Leaving directory '/build/efifs/src/EfiFs-arm'
```

All other architectures build fine in version 1.10. Only `arm` fails for it.

Version 1.9 builds fine for all architectures with the same compiler/system, including `arm`.

System information:
  - OS: Arch Linux
  - arm-none-eabi-gcc: 14.1.0

pbatard, 2024-06-06:
That's a GRUB upstream issue. Their declaration is not consistent with the actual function definition.

I suggest you send a patch to GRUB upstream, as it will naturally find its way down to EfiFs since, unless I do experience issues in my own builds (and I'm afraid I don't use Arch for that) I don't have the time to go around and patch more of GRUB than I currently do, and again, this is a pure GRUB issue that should really be fixed upstream.

dbermond, 2024-06-29:
Thank you for the reply and clarification about where the problem is coming from.

I could fix the issue by patching grub. More build errors appeared after fixing this one, but I could also patch them.

It was also needed to pach gnu-efi to disable `-Werror` on it.

For reference, and in the case you maybe you want to pick them, the patches are available here (applies after your patch):
  - [grub patch 01](https://gitlab.archlinux.org/archlinux/packaging/packages/efifs/-/blob/1.10-1/010-efifs-grub-fix-conflicting-parameter-types.patch?ref_type=tags)
  - [grub patch 02](https://gitlab.archlinux.org/archlinux/packaging/packages/efifs/-/blob/1.10-1/020-efifs-grub-fix-incompatible-pointer-types.patch?ref_type=tags)
  - [gnu-efi patch 01](https://gitlab.archlinux.org/archlinux/packaging/packages/efifs/-/blob/1.10-1/030-efifs-gnu-efi-remove-werror.patch?ref_type=tags)

Fell free to close this issue if you want.

closed 2024-11-29 by pbatard