pbatard/EfiFs #48: does QEMU UEFI need driver for SD card?

bam80, 2025-05-07:
Hi, maybe you could help me to understand if UEFI driver is needed for this, or the problem is with something else:
https://gitlab.com/qemu-project/qemu/-/issues/2954
https://github.com/tianocore/tianocore.github.io/issues/88

pbatard, 2025-05-07:
Pretty sure you need a UEFI SD card driver. It is not provided as standard by UEFI. I think you are assuming that SD card drivers are standardized, as is the case with USB 3 and xHCI controllers, but that is not the case.

For instance we had to write one for the Raspberry Pi: https://github.com/tianocore/edk2-platforms/tree/master/Platform/RaspberryPi/Drivers/SdHostDxe. So, QEMU, just like real hardware, will need a UEFI driver for the non standardized peripherals it uses, such as SD card readers.

closed 2025-05-07 by pbatard

bam80, 2025-05-07:
Thanks, isn't it the driver required?
https://github.com/tianocore/edk2/blob/master/MdeModulePkg/Bus/Pci/SdMmcPciHcDxe/SdMmcPciHcDxe.c

Apparently, someone already reported the problem, but there was no response:
https://edk2.groups.io/g/bugs/topic/bug_1632_new_sdmmcpcihcdxe/34456908

pbatard, 2025-05-07:
Dude, I'm not going to do your work for you, sorry. You asked a question and I answered it to the best of my knowledge. If you need more than this, please see with the QEMU folks or on the QEMU support forums, since this should never have been raised as an EfiFs support request in the first place.